SevenBits/Mac-Linux-USB-Loader #130: USB not appearing in "Choose Destination" option

ajmichaels, 2016-08-31:
**Short Description of Problem:** USB is not appearing as an option, even though I can use it fine

**Expected Behavior:** 

USB appears as an option just like the tutorial screens

**Actual Behavior:** 

USB is not appearing as an option - nothing is appearing.  The "Choose Destination" box is blank.

**Steps to Reproduce:** 

Not sure - This happened first try.

[image]

SevenBits, 2016-09-01:
@ajmichaels Have you read [the FAQs](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Frequently-Asked-Questions#3-why-isnt-my-usb-drive-detected-it-should-be-formatted-as-fat)?

ajmichaels, 2016-09-01:
@SevenBits Yes, I've tried formatting it as every available option disk utility provides and it's not being detected.

SevenBits, 2016-09-01:
@ajmichaels Please try enabling the 'Detect Hard Drives' option in the preferences.

ajmichaels, 2016-09-01:
@SevenBits Tried that before and got it working, thank you :) If I may ask one more question, I've followed the instructions here http://www.howtogeek.com/213396/how-to-boot-a-linux-live-usb-drive-on-your-mac/ to get persistence working with your app, but the drive is once again not showing up anywhere (booting into ubuntu is working).  I didn't see anything in the FAQ about persistence with the GUI option.  Is it that I need a second USB?

closed 2016-09-01 by ajmichaels