SevenBits/Mac-Linux-USB-Loader #131: Persistence Manager

clayauld, 2016-09-02:
**Short Description of Problem:** The persistence manager seems to work properly for a while then suddenly crashes the entire app. After that happens the persistence file seems to be the proper size.

**Expected Behavior:** 

I would assume the expected behavior should exhibit some sort of proper exit of the program where confirmation or errors are presented showing the results.

**Actual Behavior:** 

The app just crashes when nearing the end of the persistence setup.

**Steps to Reproduce:** 

This happens every time I try to do any persistence file setup.
[error log.pdf](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/451573/error.log.pdf)

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2016-09-06:
@clayauld Will close unless I hear more from you.

clayauld, 2016-09-06:
Go ahead and close it.

closed 2016-09-07 by SevenBits