SevenBits/Mac-Linux-USB-Loader #132: Kali Linux usb persistence unable to mount and therefore not working

oproz, 2016-09-03:
**Short Description of Problem:** 
Hello,
I used mac linux USB loader 3.3 to create a persistence partition on my usb stick but when booting it fail to mount the partition

**Expected Behavior:** 
During the boot process it should mount my persistent data

**Actual Behavior:** 
I get this mesage instead : 
`mounting /dev/sdb2 on /live/medium failed : invalid arguments`
`mounting /dev/sdb2/persistence on /live/medium failed : invalid arguments`

**Steps to Reproduce:** 
Create a persistent partition via Mac linux USB loader and boot I cant supply screenshots since it is during the booting process

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2016-09-03:
@oproz Hi there, just to clarify: Mac Linux USB Loader’s built-in persistence tool only works with Ubuntu-based distributions. Kali requires separate steps which are detailed in this video (which I am linking to for informational purposes and do not endorse): https://www.youtube.com/watch?v=mDRbTHCoj8U

SevenBits, 2016-09-06:
@oproz Will close unless I receive more input from you.

closed 2016-09-08 by SevenBits