SevenBits/Mac-Linux-USB-Loader #133: Mac Linux USB Loader not working on my new Mac Book Pro

uxkenchris, 2016-09-04:
I downloaded the USB Loader program, it is not seeing my usb device.  USB Boot Controls and advanced configuration is greyed out.  My usb thumb drive is reformatted Master boot disc, should it be GUID?  it did not find that either.  How do i get it to show??

**Expected Behavior:**  

I should be able to select my usb drive

**Actual Behavior:** 

the installer cant find it, but disk utility and my desktop shows it

[image]
[image]
[image]

**Steps to Reproduce:** 

put text here

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2016-09-04:
@uxkenchris Do you have the “enable hard drives” option enabled in the preferences?

uxkenchris, 2016-09-04:
No I did not. Thanks!  From a UX standpoint. It may be handy to enable it for users and to have a step by step instructions on the site.

Best!!
A UX Designer.

SevenBits, 2016-09-04:
@uxkenchris Did that solve your problem? If so, then please close out this issue.

closed 2016-09-08 by SevenBits