SevenBits/Mac-Linux-USB-Loader #134: Create Installation Media for Kali 2?

james0S, 2016-09-06:
Want to make an installer USB stick for Kali Linux 2 for Early 2015 13" MBP El Capitan...sorry if this is not a supported feature; I have been struggling with booting an installation ISO for this OS for the past 5 days using a few other methods and I'm striking out. I'm finding this a bit surprising because last February I was able to simply burn a disc, format the SSD and install Fedora. The resistance I'm getting now has me wondering if I have invested in the right computer platform : P

So far I have installed rEFInd sucessfully and have used Rufus, Unebootin, PenDrive linux and even burned a disc that works on other computers and I have not been able to boot this software on my Mac. The only thing that has worked is Mac-Linux-USB-Loader, but I can only find ways to make Live USB sticks so far, and these jump straight to the desktop without giving installation options. 

Is is possible to edit the Enterprise file to point at the "install" folder for boot rather than "live"? I have tried a few things and totally failed.

SevenBits, 2016-09-06:
@james0S This isn't a supported configuration, but it might be doable. What have you tried so far?

> So far I have installed rEFInd sucessfully and have used Rufus, Unebootin, PenDrive linux and even burned a disc that works on other computers and I have not been able to boot this software on my Mac. The only thing that has worked is Mac-Linux-USB-Loader...

Believe me when I say that this is a common sentiment. ;)

pufferaway, 2016-09-08:
I've run into the same issue since Kali changed to Debian's latest installation process. It seems the "syslinux" boot loader that provides a grub menu is bypassed, which is what you should see initially when booting into Live Kali or Debian. Using your installer and rEFInd, Kali will boot directly into the Live desktop on my 2008 iMac running 10.8.5. Unfortunately, Kali no longer has any installation options available once you reach the desktop... you can only launch the install from the grub menu. I have tried "dd" direct to USB without success... I have the copied files from that installation here if it would save you any troubleshooting time: https://app.sugarsync.com/iris/wf/D2805917_86452621_646578
Most likely this can be traced directly to Debian, as I believe Kali is following their building process. Just FYI... CentOS installed just fine!
If I can assist with any testing, just let me know!

SevenBits, 2016-09-08:
@pufferaway I don’t have a Kali ISO in front of me so I don’t know this for sure, but perhaps you can just change the `kernel` and `ignited`v parameters in the Enterprise configuration feel to point to the installer rather than the live desktop?

pufferaway, 2016-09-08:
I'll give it a shot and update you.

james0S, 2016-09-08:
I was not sure what to do so my efforts in editing Enterprise were most likely clumsy... I have been trying some other methods like those outlined here:

https://www.youtube.com/watch?v=D1YohWypLSo

This ended up getting stuck on detecting the CD-ROM drive after booting Debian installer for me. Tried using an actual burned disc for the installer to use but it tried to use the disc and failed. 

: P 

Oh well If anyone has any luck I would be thrilled to hear it; I have been distracted from this by work and parenthood. Mac-Linux-USB-Loader was worth my $5 USD even without this feature.

alphaswell, 2016-09-08:
I had the same problem last night and gave up after a few hours. I'm trying again this morning, I've managed to change the kernel it loads to /install/vmlinuz but even with the line

initrd /install/initrd.gz

it's still loading the regular initrd.img from the boot directory. it still boots fine with these parameters, but it boots straight to the desktop environment still. there's a file called 'g2ldr' in the root directory of the iso which probably loads the initial menu, but I have no idea how I'd get enterprise to load that first.

I'll try a few other configurations along these lines but any more specific instructions as to what the enterprise.cfg might look like would be incredibly helpful.

SevenBits, 2016-09-08:
@alphaswell Could you post your complete configuration file?

alphaswell, 2016-09-08:
#This file is machine generated. Do not modify it unless you know what you are doing.

entry Kali
family Debian
kernel /install/vmlinuz initrd /install/initrd.gz findiso=/efi/boot/boot.iso boot=install noconfig=sudo username=root hostname=kali

even changing the boot to install like this doesn't change the ramdisc it loads.

alphaswell, 2016-09-08:
i also tried using 

ramdisk /install/initrd.gz

and

initrd=/install/initrd.gz

but it still boots /live/initrd.img every time

alphaswell, 2016-09-08:
#This file is machine generated. Do not modify it unless you know what you are doing.

entry Kali
family Debian
initrd /install/initrd.gz kernel /install/vmlinuz findiso=/efi/boot/boot.iso boot=install noconfig=sudo username=root hostname=kali

this booted the right ramdisk but the wrong kernel. i'm going to try keeping both those instructions on their own line.

alphaswell, 2016-09-08:
#This file is machine generated. Do not modify it unless you know what you are doing.

entry Kali
family Debian
initrd /install/initrd.gz 
kernel /install/vmlinuz findiso=/efi/boot/boot.iso boot=install noconfig=sudo username=root hostname=kali

@james0S @SevenBits  this one booted the right kernel and ramdisk and now the install script is running.

alphaswell, 2016-09-08:
however, this is the debian install script, as far as i can tell not the kali install script, and it fails as it tries to load the necessary files from the cd-rom drive. i think i'm officially out of ideas but i'll keep looking at it.

alphaswell, 2016-09-08:
#This file is machine generated. Do not modify it unless you know what you are doing.

entry Kali
family Debian
initrd /install/gtk/initrd.gz 
kernel /install/gtk/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

this loads the kali graphical installer but it's still trying to do it from the cdrom and I can't find a way around it. sorry for flooding this thread.

james0S, 2016-09-08:
Hey, no problem at all... I'm trying to use your work to install Debian, which be converted to Kali using this script:

http://www.etcg.pw/downloads/debian-to-kali.bash

I will let you know if I have any success. I ran into the same CD-ROM install issue with Kali as well using etcg's method, btw.

james0S, 2016-09-09:
FYI everything fails even with a CD-ROM/DVD Drive. It seems that it can find the files on the disc but it can't use them. Some kind of I/O problem perhaps.

SevenBits, 2016-09-09:
@james0S If you can boot into the live desktop, I’m not sure why there’d be a problem booting the installer...

alphaswell, 2016-09-09:
It automatically boots into the live desktop which seems to work perfectly, but I don't see an option to load the installer anywhere once inside.

james0S, 2016-09-09:
I think what SevenBits means is that "if you can boot to live, then why can't you boot to the installer?" 

Yeah I don't get it either. Speculating will lay my ignorance bare, so it is what it is for now. I can make a partition on the SSD a Live partition with persistence, however it's not really the same thing.

SevenBits, 2016-09-09:
@alphaswell @james0S My point was that it is highly improbable that the live desktop and the installer use different kernels or have different system requirements. Thus, if your system can boot the live desktop, then it should be able to boot the installer (which surely has lower system requirements than the live desktop).

Really, I blame Debian. What could possibly benefit from having the installer be not merely a separate boot setting, but instead a whole other kernel and initrd. They probably have a good reason, but it does seem to be a feature without a benefit AFAIK, especially since it complicates tools like Enterprise, my boot loader, and, thus by extension, Mac Linux USB Loader.

alphaswell, 2016-09-09:
Yeah, I was going to say that it's 100% the fault of the distro, it makes no sense. Today I'm going to experiment with live building a kali iso on a virtual machine running on my main computer. Maybe there's a way to remove the live desktop environment from the iso and leave just the installer that would work.

alphaswell, 2016-09-09:
the other option would be to figure out how to get Enterprise to load the default grub menu of the iso which probably shouldn't be that hard, but is completely beyond my current knowledge.

SevenBits, 2016-09-09:
@alphaswell Enterprise doesn’t have a feature to take you to the distro’s menu, but that’s a feature, not a bug... when I started this project four years ago, most distributions had no EFI support or just flat out would not load, so it was pointless to use the distribution’s settings to setup the boot since they wouldn’t have worked anyway.

Didn’t you say that this configuration worked?

    entry Kali Installer
    family Debian
    initrd /install/gtk/initrd.gz 
    kernel /install/gtk/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

BTW, you are not likely to be successful with a DVD. CDs boot using [El Torito](https://en.wikipedia.org/wiki/El_Torito_(CD-ROM_standard)), but Macs require EFI. Unlike with a USB, getting a CD/DVD to support both EFI and El Torito is no easy task. I believe even Canonical had a problem with it.

alphaswell, 2016-09-09:
Yeah, that configuration worked to get into the graphical install screen, but around step 5 or 6 it starts looking for the files in the CD-ROM drive where they can't be found. My optical drive in my MacBook has been broken for many years now so there was never a hope for me doing it that way anyway.

denismalofeev77, 2016-11-22:
I have a macbook pro early 2008. Same problem. You got something?

SevenBits, 2016-11-22:
@denismalofeev77 Did you try the above configuration from September 9th?

denismalofeev77, 2016-11-22:
entry Kali Installer
family Debian
initrd /install/gtk/initrd.gz 
kernel /install/gtk/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

When using this configuration, there is LIVE mode and immediately begins the installation. During installation, requested CD and the installation continues from the CD. When markup HDD partitions are not there. I then looked through the Disk Utility. Can not install GRUB.

[image]
[image]
[image]
[image]

denismalofeev77, 2016-11-22:
I downloaded the latest version of Kali. http://cdimage.kali.org/kali-images/kali-weekly/kali-linux-2016-W47-amd64.iso installation was successful. However, when running on a hanging position.

denismalofeev77, 2016-11-22:
[image]