SevenBits/Mac-Linux-USB-Loader #135: The archive "mac_linux_usb_loader.xip" does not com from Apple.

brianwalborn, 2016-09-10:
**Short Description of Problem:** Upon attempting to unzip the .xip file that was purchased, I'm getting the following system error: "The archive 'mac_linux_usb_loader.xip' does not com from Apple." I also attempted to use StuffIt Expander, but to no avail.

**Expected Behavior:** The file to unzip successfully.

**Actual Behavior:** Receive a system error.

**Steps to Reproduce:** Purchase and download the Mac-Linux USB Loader and attempt to unzip file.

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!
[image]

SevenBits, 2016-09-11:
@brianwalborn What version of OS X are you running?

brianwalborn, 2016-09-12:
I'm on the 10.12 Sierra beta. Do you plan on adding support for it in the near future?

SevenBits, 2016-09-12:
@brianwalborn Apple changed its plans on short notice, and now will not extract XIP files from third party developers. They previously recommended that developers use xip files. For the time being, please use a third-party app (like 'The Unarchiver' on the Mac App Store, it's free) to do the extraction. I will post a link to this issue report on the website.

closed 2016-09-12 by SevenBits