SevenBits/Mac-Linux-USB-Loader #137: Can't Get to Graphical Install Option In Kali

Addisonbean, 2016-09-22:
I am trying to install Kali on a partition on my Mac, so I decided to use Mac Linux USB Loader to install it. However, I can't get to the menu as shown in this photo:
[image]

It seems like the menu is just skipped entirely. How can I get to that menu, so I can install it onto a partition?

SevenBits, 2016-09-22:
Mac Linux USB Loader is built to skip the distribution’s own boot menu; its boot loader loads the kernel itself. The reason for this is that most distributions didn’t use settings that would work on Macs.

This strategy did work, but then Debian came up with a new installation method whereby the installer is a totally separate kernel and ram disc from the live boot. I have no clue why they did this, but they did. I’m not sure how to handle this yet.

Addisonbean, 2016-09-22:
Huh, that's strange. So if I got an older Kali iso that uses the older installation method, should it work?

Addisonbean, 2016-09-23:
Or is there a way to install Kali on a partition while running Kali?

SevenBits, 2016-09-24:
> Huh, that's strange. So if I got an older Kali iso that uses the older installation method, should it work?

I’m assuming so, though I’m not sure how far back you’d need to go.

> Or is there a way to install Kali on a partition while running Kali?

You can’t install Kali unless you boot into the installer, which Mac Linux USB Loader does not do by default. You can modify the configuration file to do this, though.

SevenBits, 2016-09-24:
Closing as this is a duplicate of #134.

closed 2016-09-24 by SevenBits

lo0oper, 2018-11-19:
i am using dell 5577 with nvidia 1050 graphic card i don't get the option of the graphical install so i opted for the install graphically in the advance options ..
will this work?