SevenBits/Mac-Linux-USB-Loader #142: Cannot boot Ubuntu on Mac: "Need to load the kernal first"

Timbo879, 2016-11-02:
**Short Description of Problem:** I cannot boot Ubuntu on my Mac. I am getting an error: 

> Loading initial RAM disc...error: you need to load the kernal first.

After I receive this message, I am show another message:

> Could not boot the Linux distribution. Perhaps your drive could not be read by GRUB. Or, maybe the distribution organizes its files in a way that's not supported yet. Please email the developer at sevenbitstech@gmail.com to report this!

I am running this on a Mac, Early 2011 with macOS Sierra 10.12

Any tips on how I can fix this?

Thanks!

SevenBits, 2016-11-02:
Hi there @Timbo879,

Could you please post the configuration file that was written on your USB? To get your configuration file, open the Setup USB Device panel, select your USB, and then press Edit Enterprise Configuration File. TextEdit will open. Paste its contents here.

Thank you!

Timbo879, 2016-11-02:
Thank you @SevenBits!

Here is my configuration file:

#This file is machine generated. Do not modify it unless you know what you are doing.

autoboot 0
entry Ubuntu
family Ubuntu

SevenBits, 2016-11-02:
@Timbo879 What options did you select when creating the USB?

Timbo879, 2016-11-02:
@SevenBits The only option I selected was `Skip the boot selection menu`

curiouscurious, 2016-11-03:
I am having the same issue with the same error messages. The only difference is I'm on a Macbook Air early 2014, OS X 10.9.5. While creating the USB, I first set aside 1 GB using the slider on the persistence manager, then I "Created Live USB", chose my USB, and it auto-selected "skip the boot selection menu". Since I'm installing the newest release of Kali Linux, I didn't check any of the other boxes. My USB is also formatted MS-DOS (FAT). My configuration file is pasted below:

#This file is machine generated. Do not modify it unless you know what you are doing.

autoboot 0
entry Kali
family Debian
kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

curiouscurious, 2016-11-13:
I started from scratch, redid everything at it works great. Not sure why it didn't work the first time, but thank you

Timbo879, 2016-11-13:
Glad to hear! I am still trying to figure my issue out.

SevenBits, 2016-11-30:
@Timbo879 Are you still having this issue?

Timbo879, 2016-11-30:
@SevenBits Yes I am, unfortunately. Any suggestions?

SevenBits, 2016-11-30:
@Timbo879 Are you trying to boot Ubuntu, or Kali Linux?

Timbo879, 2016-11-30:
@SevenBits Ubuntu

Timbo879, 2016-11-30:
@SevenBits I got a different error when I tried to boot Kali

smizacks02, 2016-12-06:
Hi @SevenBits I'm sharing this error too.
Here's my config file:
`#This file is machine generated. Do not modify it unless you know what you are doing.

entry Ubuntu
family Ubuntu
`

Attempting to boot on a MacBook Pro running OS X 10.12.1
I formatted the USB drive in FAT32 and MBR.
Used the Linux Mint 18 distort (linuxmint-18-cinnamon-64bit.iso)

Are there advanced boot options I must enable?
Or do I need to configure Enterprise config file pointing to a specific kernel location?

Thank you for a great tool and I'm confident I can get it to work.

SevenBits, 2016-12-07:
@smizacks02 Are you receiving the error described at the top of this page?

smizacks02, 2016-12-07:
Yes, only the first error (about needing to load the kernel first).
Am I making an elementary error somewhere in how I'm using Mac Linux USB Loader?
I leave both options unchecked (regarding EFI kernel and pre-Ubuntu 14.10 files).

SevenBits, 2016-12-07:
@smizacks02 If you are using the latest Ubuntu then you should stick to the defaults established by Mac Linux USB Loader.

SevenBits, 2016-12-19:
Closing due to lack of activity. If this is still an issue, please re-open and add additional information.

closed 2016-12-19 by SevenBits