SevenBits/Mac-Linux-USB-Loader #144: Won't find USB drive(s) on MacOS Sierra

forrie, 2016-11-03:
**Short Description of Problem:  I have two USB drives inserted into my Mac Pro cylinder that this program won't locate.

**Expected Behavior:** 

I should be able to select which USB to install the selected ISO on

**Actual Behavior:** 

There are no devices shown

**Steps to Reproduce:** 

I have two USBs mounted and verfied on the system; when I run this application and try to set up the ISO onto a device, there is nothing listed to select from.

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2016-11-13:
Have you checked whether or not the "enable hard drives" option is enabled?

SevenBits, 2016-11-20:
Closing due to lack of new activity.

closed 2016-11-20 by SevenBits

TaciteOFF, 2017-03-29:
Same issue here on 10.12.4.

SevenBits, 2017-03-31:
@TaciteOFF Do you have the ‘enable hard drives’ option on?