SevenBits/Mac-Linux-USB-Loader #149: Mounting /dev/sda2 on live/medium failed: Invalid Argument

lapras08, 2016-11-16:
After successfully creating Live USB it seems my Kali Linux doesn't boot. I get this message:

[      3.654639] irq 17: nobody cared (try booting with the "irqpoll" option)
[      3.654781] handlers:
[      3.654803] [<ffffffffc016d550>] sdchci_irq [sdhci] threaded [<ffffffffc016a3c0>] sdhci_thread_irq [sdhci]
[     3.654874] DISABLING iRQ #17 
mounting /dev/sda2 on live/medium failed: Invalid Argument

Halp.

SevenBits, 2016-11-20:
@lapras08 Without a little more information, this will be rather impossible to debug.

To start off, could you please post the configuration file that was written on your USB? To get your configuration file, open the Setup USB Device panel, select your USB, and then press Edit Enterprise Configuration File. TextEdit will open. Paste its contents here.

Thank you!

lapras08, 2016-11-21:
#This file is machine generated. Do not modify it unless you know what you are doing.

autoboot 0
entry Kali
family Debian
kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

SevenBits, 2016-11-30:
@lapras08 That configuration looks okay. Maybe it’s actually booting kay and you just can’t see graphics? Try disabling auto boot if you enabled it, and then boot up with the `nomodeset` option.

jrcain3, 2016-12-18:
I am seeing the same issue. Booted with `nomodeset` with the same result.

SevenBits, 2016-12-19:
@jrcain3 What model of Mac do you have? It could just be a graphical problem manifesting as something else.

jrcain3, 2016-12-19:
@SevenBits Late 2011 15" MBP

lapras08, 2016-12-20:
I'm using the same model as jcrain3 too. Still didn't work out for me. Any other solution?

SevenBits, 2016-12-21:
@lapras08 Is that the lone Mac model with an AMD video card? AMD cards are generally not well supported in Linux.

mastover, 2016-12-30:
I have the exact same issue....same model mac as well.  Any help would be greatly appreciated.  I am trying to load Kali.  I've tried the ISO your app downloads as well as one directly from the Kali distributor, but both result in the same error.

SevenBits, 2017-04-28:
Closing this due to lack of activity.

closed 2017-04-28 by SevenBits