SevenBits/Mac-Linux-USB-Loader #151: Ubuntu-Mate bug

Dyadikos, 2016-11-29:
**Short Description of Problem:**

Boot up on a late 2009 MacBook fails over to Grub.

**Expected Behavior:** 

Smooth USB drive boot into Ubuntu-Mate

**Actual Behavior:** 

Boot up on a late 2009 MacBook fails over to Grub.

I took a look at the Mac Linux USB Loader tool, it looks like it uses its own UEFI boot loader to boot into grub which can then mount the distribution's ISO and continue booting Linux. A common problem that users seem to face with this tool (as detailed at https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/21) is that the tool uses a preconfigured instance of grub which expects each distribution to be laid out in a certain way. If the ISO is not named as it expects or if the files within the ISO do not match the layout expected by the tool, grub will fail to boot into Linux and dump you into the prompt that you see.

You can start by checking to see if the UbuntuMate ISO on the USB has been named boot.iso and is in the /EFI/BOOT/ directory. If you still have the same problem after that you might need to see about creating a custom grub.conf to drop in alongside the ISO. For that I'd recommend poking the creator of the MLUL tool.

**Steps to Reproduce:** 

Install Ubuntu-Mate on a USB drive, using the Max Linux USB Loader, then try to boot using a late 2009 MacBook.

SevenBits, 2016-11-30:
@Dyadikos Thank you for the bug report. I should firstly note that #21 was an issue that was filed before the introduction of my boot loader [Enterprise][1], which is vastly more configurable than the hard-coded approach I was using before. Enterprise has been used since December 2013. Of particular note, Enterprise can boot any Linux ISO that supports _loopback_ (i.e., finding its root file system inside of an ISO file).

Enterprise is very configurable. You can specify the kernel location, any required boot parameters, and more. You can download Enterprise binaries and read its documentation at [its web site][1]; it is actually very easy to set up if you have a bit of know-how. If not, Mac Linux USB Loader can do all of the work for you.

To address your issue with Ubuntu MATE, well, I haven’t tried it, but if it uses the same ISO layout as vanilla Ubuntu, then Mac Linux USB Loader’s default settings for Ubuntu should work fine. If not, then you can easily tweak the Enterprise configuration file to point the kernel to the right place. Hope this helps.

[1]: http://blog.sevenbits.io/Enterprise/ "Enterprise"

SevenBits, 2016-12-19:
Closing due to lack of activity.

closed 2016-12-19 by SevenBits