SevenBits/Mac-Linux-USB-Loader #152: Where to install persistence

stuzog, 2016-12-01:
**Short Description of Problem:** When adding persistence to a USB stick, the recommended casper-rw installation location is unclear.

**Expected Behavior:** 

The Persistence installer should not allow installation in an unacceptable location and documentation should explain where this could be installed if optional locations are available.

**Actual Behavior:** 

When adding persistence to a USB stick, the user interface apparently allows installation at root or inside a listed directory. This is confusing for new Linux users -- where should it go?

**Steps to Reproduce:** 

Existing user interface for adding persistence.

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2016-12-03:
@stuzog Thank you for the suggestion!

TacoByte, 2016-12-23:
On this topic, so where is the appropriate place to install persistence? Is the root fine? Cause I've been trying to make persistence work with no success. I'm using a 16GB USB drive so maybe it's too small.

SevenBits, 2016-12-23:
@TacoByte For Ubuntu, it should be at the root of the USB, named `casper-rw`.

TacoByte, 2016-12-23:
OK thanks for the quick reply! I will try with a larger USB drive and create another issue if I still can't get it to work

SevenBits, 2017-01-25:
There is now documentation on this in the wiki so I’m closing this thread.

closed 2017-01-25 by SevenBits