SevenBits/Mac-Linux-USB-Loader #153: Distribution Downloader Doesn't Tell You Where It's Saving The Files Too.

Dygear, 2016-12-06:
**Short Description of Problem:**
Distribution downloader does not show where the files are being downloaded too. I can't find the two ISO files that it downloaded, if indeed it downloaded anything at all. I don't know where these files went. They are not in my downloads dir, they are not in my root, they at not in my file system at all it seems.

**Expected Behavior:** 
Letting me select where to save the ISO file too, or having the installer open to the same location where it downloaded the ISO files too.

**Actual Behavior:** 
No freaking idea where the files I downloaded went. None. Zero. Completely useless.

**Steps to Reproduce:** 
Literally use the program. Forgive that this report is terse, this is an extremely annoying issue.

[image]

Somewhere in there it should show you where the files are saving too.

SevenBits, 2016-12-07:
It is possible to show the files in Finder by right-clicking the distribution in the left-hand panel and selecting “Show in Finder”.

Dygear, 2016-12-12:
Please add that as a button next to `Download This Distro` as that is non-obvious.

SevenBits, 2016-12-14:
@Dygear Adding it as a button would probably not look good in the UI. I will see if I can find a better way to communicate this information to the user.

commit a35a28ca (2016-12-14): Fixes #153

Adds a small label and user showing the user where downloaded ISOs have
been saved to.
closed 2016-12-14 by SevenBits