SevenBits/Mac-Linux-USB-Loader #154: "Grant Access" does not work as expected. Appears as a "File Open" dialog.

JoeDotCom1, 2016-12-14:
**Short Description of Problem:** 
After Clicking "Perform Installation" the "grant permission" dialog is displaying as a "file open" window, clicking "Grant Access" does not work as expected. 

*I'm running OS 10.9.5, and just downloaded the most recent version of your app a few min ago. 

thanks for your time.

**Expected Behavior:** 
I believe this should be a permissions dialog, not a "file browsing" dialog. 

**Actual Behavior:** 

screenshot:
[image]

SevenBits, 2016-12-14:
@JoeDotCom1 When the file browser appears, make sure you are viewing the contents of your USB drive. If you press Grant Access anywhere else the software will reject you.

I know this is inconvenient, but it is necessary to comply with Apple’s sandboxing rules.

SevenBits, 2016-12-19:
Closing due to lack of activity.

closed 2016-12-19 by SevenBits