SevenBits/Mac-Linux-USB-Loader #155: Persistence Manager doesn't show my USB

dikrek, 2016-12-19:
**Short Description of Problem:** 

Just created a bootable USB with your program (great effort and thanks for making this!)

Persistence isn't part of the creation process but a separate option. So I went to persistence after setting up the USB with Ubuntu 16.10.

Nothing shows up in the list for "target USB" - just three dashes even though the drive is mounted and the program successfully built Ubuntu on it.

**Expected Behavior:** 

create persistence

**Actual Behavior:** 

doesn't create persistence

**Steps to Reproduce:** 

click on persistence setup

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

dikrek, 2016-12-19:
I used a larger USB and it DID work with persistence manager. Is there a size thing?

The one it didn't work on had 3GB free though.

SevenBits, 2016-12-19:
@dikrek Does your USB’s name have a space inside of it? Unfortunately, due to system limitations, persistence cannot be setup correctly on devices with a space in their name.

dikrek, 2016-12-19:
it was the space, thanks. I removed the space and I can create the device. On the other device where I created a 2GB space plus used Ubuntu 16.04, I don't see that space mounted anywhere when I start the OS, how is the persistence used? Or is it just automatic?

SevenBits, 2016-12-19:
@dikrek The persistence is not seen as an external drive; rather, it is superimposed on top of the read only filesystem on the ISO. If you make a change, it will be automatically saved and will appear again when you boot with persistence enabled.

closed 2016-12-23 by SevenBits