SevenBits/Mac-Linux-USB-Loader #156: Add some Rescue CD ?

GGGunrunner, 2016-12-23:
Hello and thanks for this great tool!

Is there any possibility to add some Rescue CD with memtest, gparted, various backup and repair utilities?

Would be great!

SevenBits, 2016-12-23:
@GGGunrunner This is something I've considered, but they won't make it for the current version, which will be released either today or tomorrow.

GGGunrunner, 2016-12-24:
Thanks!

Is it at least possible to slip Memtest86 iso to the loader?

I briefly looked into how its organized. It has has a binary in /isolinux/memtest
Maybe its possible to point the execution to it instead of vmlinuz?

Memtest does support UEFI, but I'm not sure how closely to the Apple's implementation.

SevenBits, 2016-12-24:
@GGGunrunner You can certainly try to get it to work yourself by manually updating the configuration file; however, I am not going to support it officially for the current release. Maybe in the future.