SevenBits/Mac-Linux-USB-Loader #157: Does not load the ISO to the USB stick

obscurion, 2016-12-29:
The app copies the efi folder to the usb stick but nothing else

**Expected Behavior:** 

Copy the iso to the usb stick

**Actual Behavior:** 

Nothing but the efi folder is copied to my usb stick. it doesn't seem to matter which distribution I use. I've tried Ubuntu 16.10 and Mint 18.1 Cinnamon. I've pre-formatted the usb stick in several different ways. No luck. The app ends its process with "Success!" so it thinks it's done its job.

**Steps to Reproduce:** 

Follow the on screen directions'

If it helps: I'm using a late 2009 iMac running 10.12.2 (16C67)

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2017-01-14:
Hi @obscurion, nothing should be copied to the root of the USB drive other than an `efi` folder. Can you boot the USB successfully?

obscurion, 2017-01-14:
> 
> Hi @obscurion <https://github.com/obscurion>, nothing should be copied to the root of the USB drive other than an efi folder. Can you boot the USB successfully?
OK, that’s a bit different than the Fedora tool. I’ve successfully used their's to load up Fedora distributions.

I’m not sure what is wrong when using your tool. Sometimes the usb stick doesn’t show up on the boot list. I got it to show once but when chosen the screen went black. I think this may have been a (lack of) driver issue.

I haven’t tried it for a few days and will try again.

Thanks for your response.

David

closed 2017-01-25 by SevenBits