SevenBits/Mac-Linux-USB-Loader #158: Not Working Kali Boot on MBP 2016 15" (w/ Touchbar)

miki2o, 2017-01-11:
**Short Description of Problem:** Hi, I've compiled from scratch using Xcode the latest version of the loader.
When I create the live USB and load in the EFI, it loads the kernel image and the Ramdisk but then fails to boot the distro because it can't find the command "applesetos".
I've searched in the net, and I haven't found any solution, except a patch [here](https://lists.gnu.org/archive/html/grub-devel/2013-12/msg00442.html) for the GRUB.
Also, I've tried using an older version of the loader(3.2) and it doesn't show this error, but the pen drive deactivates(no more power) when it tries to load the distro(after kernel and Ramdisk).

**Expected Behavior:** 

Expected to boot into the grub menù for selecting install or live usb.

**Actual Behavior:** 

Dietro fails to load, after that kernel and Ramdisk get loaded correctly.

**Steps to Reproduce:** 

Compile Loader
Create Live USB
Boot from EFI
Select the option to boot kali(if autoboot isn't activated)
Then you see the error.

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2017-01-14:
@miki2o Even if it can't find the command, the system should still boot. Do you perhaps have a graphics issue?

miki2o, 2017-01-14:
But it doesn't boot after the error and i don't understand why...

miki2o, 2017-01-14:
http://imgur.com/QLi88vB
Here is a photo of the error...and the pendrive is shut down when the error appears
I've tried both with and without nomodeset.. the result is the same

miki2o, 2017-01-14:
Okay so I've tried on a MBP 13" 2015 of my friend and it's showing the error "applesetos" but then it boots....so I can assume that the problem is on the new MacBook Pros 2016

LVLAaron, 2017-01-15:
https://www.sevenbits.io/support.html#!/enterprise-linux-uefi/development-feature-reque:error-cant-find-command

SevenBits is saying it's something up with the distributions. I have a USB that works with a Macbook Pro and does not work (does the same as yours) on a 2016 Macbook

miki2o, 2017-01-15:
So it's not a problem of the loader?

miki2o, 2017-01-15:
I round this... https://www.reddit.com/r/Ubuntu/comments/5c8s1k/warning_2016_macbook_pro_is_not_compatible_with/

Gonna try it later

LVLAaron, 2017-01-15:
I'm not smart enough w/ Mac stuff to know.

miki2o, 2017-01-15:
Well i tried booting with the command intremap=nosid and it's actually booting but nothing(keyboard,touchpad,network) is working except the display...so i can assume it's a problem of linux distros with drivers/cfgs and not the loader

LVLAaron, 2017-01-15:
Bummer. Thanks for trying. Arch has some instructions for installing on mac... I'm trying those.

SevenBits, 2017-01-15:
> So it's not a problem of the loader?

No, the distributions have to update their drivers, unfortunately.

miki2o, 2017-01-15:
Mmmh okay. However i tried booting ubuntu with that command and it's booting
Then i tried booting kali with that command and is not booting.
Can this be related to the loader?

Shannon713, 2017-01-15:
The same thing happened to me, but I have a Mac from late 2013. Is it just too old or is there a way to boot Kali? (Sorry i am new to this)

SiddhaKilaru, 2017-01-20:
I have the same problem with the 13" touch bar.... i have countless hours trying to get kali work

Shannon713, 2017-01-20:
yes same. Using other software i managed to get Kali live working, but i cant add persistence :|

defatpm, 2017-01-26:
Same issue here. 

What did you use to get Live working?

Shannon713, 2017-02-03:
I just followed the instructions on the official site. It used rEFInd boot manager, and i imaged the iso using mac terminal

miki2o, 2017-02-03:
I can't manager to installato rEFInd on my MBP 15" w/touchbar
How did you install it?

railbender, 2017-02-10:
+1  
I have a Macbook Pro with the touch bar and I am experiencing the same problem and same "applesetos" message when trying to boot to a Kali image.

SevenBits, 2017-02-10:
@railbender The issue is related to your Apple touch bar undoubtedly. The `applesetos` message is an oversight from me, but won’t affect the boot process.

KAYJEF, 2017-02-19:
Same issue with me and I have iMac Late 2014 5K so it isn't the touchbar

whopapa, 2017-02-22:
Hi there,
I ran into the same issue. This is the only page I found other people who ran into the same issue. Therefore I was wondering whether there was a solution available yet, or that I just have to wait for the distributions to be updated. (I also have a MBP with touchbar)

SevenBits, 2017-02-23:
@whopapa There is nothing I can do. The individual Linux distributions are responsible for maintaining their own drivers.

railbender, 2017-02-23:
It would be ethical to put a note in your sales material that you know ithis software doesn't work with the latest MacBook Pro products until the drivers get updated.

SevenBits, 2017-02-23:
Hi @railbender, there already is a message about this on the [system requirements page](https://www.sevenbits.io/mlul/sys-requir.html).

KAYJEF, 2017-02-23:
Hi
   I have an imac late 2014 5k and i have issues with cant boot so it isnt the touchbar.

pgilfernandez, 2017-03-19:
Hi,

Same problem here with a MBP 15'' late 2011 so the "applesetos" is not only related with the MBP with touchbar...

Thanks

SevenBits, 2017-03-21:
@KAYJEF @pgilfernandez Have you both tried booting with the `nomodeset` option, under the Advanced Boot Options menu of the boot loader? If graphics don’t show up this is likely why.

Yes, I realize the “applesetos” thing is annoying, but it doesn’t actually affect the boot process. I’ll fix it in the next release.

LVLAaron, 2017-03-21:
I have tried that with no success (current macbook proper)

KAYJEF, 2017-03-21:
I will try this today hopefully 

Ken

pgilfernandez, 2017-03-22:
Hi,
I have tried the "nomodeset" option but it still shows the "applesetos" error... anyway it goes a little forward showing new messages/errors:

> [8.455007] [drm:radeon_init [radeon]] *ERROR* No UMS support in radeon module!
> [8.572658] scsi host6: runtime PM trying to activate child device host6 but parent (2-1.2:1.0) is not active

Then it shows the Ubuntu logo (I'm trying to boot 16.10 version) and I can still read:

> [16.296146] [drm:radeon_init [radeon]] *ERROR* No UMS support in radeon module!

Finally it shows all messages/errors:

> [8.455007] [drm:radeon_init [radeon]] *ERROR* No UMS support in radeon module!
> [8.572658] scsi host6: runtime PM trying to activate child device host6 but parent (2-1.2:1.0) is not active

> [16.296146] [drm:radeon_init [radeon]] *ERROR* No UMS support in radeon module!
> [21.437921] [drm:radeon_init [radeon]] *ERROR* No UMS support in radeon module!
> [21.669927] b43-phy0 ERROR: Firmware file "b43/ucode29_mimo.fw" not found
> [21.669961] b43-phy0 ERROR: Firmware file "b43-open/ucode29_mimo.fw" not found
> [21.669992] b43-phy0 ERROR: You must go to http://wireless.kernel.org/en/users/Drivers/b43#devicefirmware and download the correct firmware for this driver version. Please carefull read all instructions on this website.

Any idea what to do?... I couldn't not load any image...
Thanks

SevenBits, 2017-03-22:
Again, the `applesetos` thing is not something you need to worry about.

I've done some research, and it looks like you need to boot Linux with the kernel option `intremap=nosid` in order for anything to work at all. In the current version of my boot loader you will need to press 9 under the Advanced Options menu and type it very carefully.

LVLAaron, 2017-03-22:
intremap=nosid got things booted on my macbook proper. However when booted there is no keyboard/mouse. After searching it seems that this is due to drivers in the OS.

Progdjaz, 2017-03-23:
@SevenBits
Hi...
I've tried both the "nomodeset", and "intremap=nosid" boot options, I've tried them both in the same boot, both returning the same "applesetos" error, and no boot.  Any suggestions??

2016 15" MBP w/Touch Bar

LVLAaron, 2017-03-23:
When I tried the intremap setting it still displayed the applesetos bit for a few minutes and then I started to see the OS booting. Have you tried just waiting a bit?

Progdjaz, 2017-03-23:
@LVLAaron 

Maybe I'm not being patient enough...I'll try again and give it a few.  Thanks for the suggestion

Progdjaz, 2017-03-23:
@LVLAaron 

I see you mentioned some installation instructions you found a few months back now, did those instructions work for you??

LVLAaron, 2017-03-23:
Nope. Never got arch running. I didn't try very hard though.

Progdjaz, 2017-03-23:
@LVLAaron 

13 minutes of patience later, and still no boot.

I was aware of Linux boot issues with the new MBPs when they first came out.  I guess I figured it was sorted out in the 4 months since....I guess more patience may be in order

SevenBits, 2017-03-23:
@Progdjaz @LVLAaron Yes, unfortunately it may be some time before the driver issues are sorted out.

l8444, 2017-03-24:
MacBook 2016 12"
Anyone get Kali linux to boot?

@SevenBits
New to everything fyi. Tried "nomodeset", want to attempt 9: custom but freezes. Any advice?

LVLAaron, 2017-03-24:
That's odd. I have the same machine and no problem. It doesn't really matter, if you COULD boot it, you wouldn't have keyboard/mouse - Which version of Kali are you using? I can try and boot it on mine.

l8444, 2017-03-24:
"kali-linux-2016.2-amd64.iso" newest version I assume.

Interesting finds for your keyboard and mouse issue.
 https://github.com/cb22/macbook12-spi-driver
https://forums.kali.org/showthread.php?35231-Installing-Kali-to-a-Macbook-12-quot-2016-Using-USB-Drive
https://bugzilla.kernel.org/attachment.cgi?id=206671

LVLAaron, 2017-03-24:
No luck. I can get through the menus but still get stuck on applesetos. 

I give up for now. I'll be running in a VM. :/

nickmhc, 2017-03-27:
I'm having the exact same problems, but for Ubuntu 16.04. 

I have this 2011 MacBook Pro http://www.everymac.com/systems/apple/macbook_pro/specs/macbook-pro-core-i7-2.4-15-late-2011-unibody-thunderbolt-specs.html

I've found these threads, they haven't helped so far:

- https://help.ubuntu.com/community/MacBookPro8-2/Raring
- https://bugs.freedesktop.org/show_bug.cgi?id=26891#c3
- http://www.rodsbooks.com/refind/linux.html
- https://forums.kali.org/showthread.php?29611-Trouble-creating-Live-USB-Mac
- https://help.ubuntu.com/community/MacBookPro8-2/Trusty

Here's hoping `intremap=nosid` helps me get to a place where I can download the Radeon 6770 drivers I need but have no clue how to add to the install .dmg

KAYJEF, 2017-03-27:
Brief update. I get ubuntu to boot ok after a gap of time. No Bluetooth available but wired keyboard and mouse ok. Will give better update later.

SevenBits, 2017-03-27:
@nickmhc An older MacBook from 2011 should work, unless that was that one model where Apple used AMD cards instead of nVidia... can't check right now.

Unfortunately adding something to an ISO is a not beginner friendly process. If proper graphics support is a must for you (as it sounds like it is) you really need to pester Canonical about this. I certainly have, but I'm just one voice among many.

cnwl30, 2017-04-06:
hey guys. I'm relatively new to linux too, but I do know some basic scripting.

as you can see from me joining this thread. I am one of the peasants who purchased the new 2016 15 in. MacBook Pro (w Touchbar)

When I try to graphic install Kali or even just live boot it, it seems the keyboard and mouse stops working for me. I guess the new taptic keyboard is likely incompatible and unrecognizable by Kali and so far there are no mouse and keyboard drivers released.

I guess patience is key

cnwl30, 2017-04-06:
*taptic trackback my bad

chaunwleo, 2017-04-06:
Hey guys.

As you can see from me joining this thread, I am one of the peasants that purchased the new 15 in. 2016 MBP (w Touchbar). I'm relatively new to Linux but I do know some basic scripting.

When I boot the Kali installer from EFI it works fine until I get to the main screen. This goes for both Graphical Install as well as Live Boot (amd64). It seems as if the new taptic trackpad and keyboards are incompatible or unrecognized by the system. 

So far, I haven't been able to find any drivers, nor have Offensive Security released any.

I guess patience is key :/

*or u could definitely use a VM

chaunwleo, 2017-04-06:
I'm currently trying to run Kali 0.10.5 (64-bit)

The-Wolf312, 2017-05-04:
If you're trying to get the touch bar to work @Dunedan has a repo regarding to the state of Linux on Macs and that can be found here https://github.com/Dunedan/mbp-2016-linux a separate driver for Linux that allows the touch bar to be run is available here https://github.com/roadrunner2/macbook12-spi-driver/blob/touchbar-driver-hid-driver/appletb.c
You might also need to use the track pad driver but I'm sure you can find that for yourself.