SevenBits/Mac-Linux-USB-Loader #160: Error during build ld: library not found for -lJSONModel

arozwalak, 2017-01-22:
Error during build:

$ xcodebuild CODE_SIGN_IDENTITY="" CODE_SIGNING_REQUIRED=NO

ld: warning: directory not found for option '-F/Users/artur/Library/Developer/Xcode/DerivedData/Mac_Linux_USB_Loader-ftzupntfxyasrlcinvwljpuogpbh/Build/Products/Debug'
ld: library not found for -lJSONModel
clang: error: linker command failed with exit code 1 (use -v to see invocation)

MacOS 10.12.2
xcode 8.2.1

SevenBits, 2017-01-22:
Have you followed all of instructions, including setting up Cocoapods?

arozwalak, 2017-01-23:
Yes, I've installed Cocoapods and run `pod install`

SevenBits, 2017-01-25:
Maybe verify that it’s build the workspace, with the Cocoapods code included, rather than just the MLUL code?

Alternatively, you could do this from within Xcode.

ColinNg, 2017-02-13:
I'm having the same problem whether I do the build in Xcode or use the command provided by arozwalak
`$ xcodebuild CODE_SIGN_IDENTITY="" CODE_SIGNING_REQUIRED=NO`

I'm on macOS Sierra 10.12.2, Xcode 8.2.1, pod version 1.2.0, gem version 2.6.8. 

```
ld: warning: directory not found for option '-L/Users/colinng/Code/Open Source/Mac Linux USB Loader/build/Release/JSONModel'
ld: warning: directory not found for option '-L/Users/colinng/Code/Open Source/Mac Linux USB Loader/build/Release/RHPreferences'
ld: warning: directory not found for option '-F/Users/colinng/Library/Developer/Xcode/DerivedData/Mac_Linux_USB_Loader-ftzupntfxyasrlcinvwljpuogpbh/Build/Products/Debug'
ld: warning: directory not found for option '-F/Users/colinng/Code/Open Source/Mac Linux USB Loader/Pods/Sparkle'
ld: library not found for -lJSONModel
```

SevenBits, 2017-02-13:
@ColinNg You’ve run `pod install`, correct?

ColinNg, 2017-02-13:
Yes, I did! 

$ pod install
Analyzing dependencies
Downloading dependencies
Using JSONModel (1.7.0)
Using RHPreferences (1.0.0)
Generating Pods project
Integrating client project
Sending stats
Pod installation complete! There are 2 dependencies from the Podfile and 2 total pods installed.

SevenBits, 2017-02-13:
@ColinNg Try building from Xcode.

ColinNg, 2017-02-13:
I've tried both Xcode and command line, but it gives the same error. 

Think it might be because I don't have a paid Apple Developer ID? 

I also tried to confirm the settings were correct as per http://stackoverflow.com/a/26658739/2136812

Paco777, 2017-02-20:
Hi,
It looks like "Pods.xcodeproj" is not included in the "Mac Linux USB Loader.xcodeproj" project.
Adding it manually (using dran'n'drop) seems to fix the problem.

matxino, 2017-04-17:
Great Paco777 your indication solved the problem!!

closed 2017-04-28 by SevenBits