SevenBits/Mac-Linux-USB-Loader #161: How to remove grub bootloader from boot disk?

indieit, 2017-01-25:
**Short Description of Problem:**

When rebooting from a linux (or any) usb drive, even with drive unplugged, I get a grub bootloader screen. It takes a option reboot and manual boot disk selection to break out of the cycle and boot my laptop normally again.

**Expected Behavior:** 

Reboot should revert normally to my boot disk.

**Actual Behavior:** 

Reboots to a grub bootloader screen.

**Steps to Reproduce:** 

Reboot from a linux or OS X usb drive.

*Note:* It seems that grub must have gotten installed on my boot drive since this happens with the usb drive unplugged. Is there a way to delete grub, preferably without hosing my boot disk?

SevenBits, 2017-01-25:
@indieit I don’t know how GRUB would get installed on your Mac, unless you manually installed it. Did you try to install Linux?

indieit, 2017-01-25:
Hmm, I installed linux to a 2nd usb drive from the Ubuntu desktop
installer. I was very careful not to select the wrong (i.e. my boot) disk.
Yikes, I thought maybe grub was installed by the usb loader. I mounted the
efi partition on my boot drive and sure enough there are 2 directories in
there, APPLE and ubuntu (with grub in the ubuntu directory). Do you know if
it's safe to simply delete the ubuntu directory or is there more to it than
that?

indieit, 2017-01-25:
Googled it enough to feel safe and deleted the ubuntu folder. OS X still
boots so it must be ok :-) Thanks for pointing me in the right direction.

closed 2017-04-28 by SevenBits