SevenBits/Mac-Linux-USB-Loader #162: Enterprise.cfg file modification to install kali linux

ardaa, 2017-02-03:
I want to install Kali Linux to my Mac. But I can't access the GUI selection menu. It directly boots up to live version ( I know that this problem was issued before).

I have tried to reconfigure the .cfg file but I cannot modify the "boot folder:" setting. and it gives a "KERNEL PANIC" error.

If you guys could help me edit the enterprise file in order to access the GUI menu.

Thanks

SevenBits, 2017-02-03:
What do you currently have in the config file?

ardaa, 2017-02-06:
I delete the one but I have the boot thingy. 
[image]
In the photo it says boot file but I changed it to "boot folder: install

SevenBits, 2017-02-08:
Hi there,

Could you please post the configuration file that was written on your USB? To get your configuration file, open the Setup USB Device panel, select your USB, and then press Edit Enterprise Configuration File. TextEdit will open. Paste its contents here.

Thank you!

CyberTechnotrix, 2017-03-25:
how to bring boot selection menu instead of direct live boot.I want to dual boot kali with mac so i have to install it from usb boot selection menu 

My **enterprise.cfg** file looks like this-
**#This file is machine generated. Do not modify it unless you know what you are doing.

entry Kali
family Debian
kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali**

SevenBits, 2017-03-25:
@CyberTechnotrix If it is booting directly and skipping the menu, then there should be an `autoboot` entry at the top of the file. Simply remove this line.

CyberTechnotrix, 2017-03-25:
No,Its like this 
`#This file is machine generated. Do not modify it unless you know what you are doing.

entry Kali
family Debian
kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

please suggest a way to boot into boot selection menu not live  desktop

SevenBits, 2017-03-25:
@CyberTechnotrix Oh, now I see. Well, you need to add another boot entry to boot into the installer, like this:

    family Debian
    initrd /install/gtk/initrd.gz
    kernel /install/gtk/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

The current version of Mac Linux USB Loader should do this automatically.

CyberTechnotrix, 2017-03-25:
@SevenBits -Thanks,Now its loading to graphical install but gives an error:Failed to mount CD-ROM(incorrect CD-ROM/file)

Do you know,How to fix this issue?

SevenBits, 2017-03-25:
@CyberTechnotrix Hmm, I’ve heard of this happening. Kali probably thinks you’re trying to install it from a CD. Is there a way to specify some other installation source?

CyberTechnotrix, 2017-03-26:
@SevenBits ,Is there a way to boot to boot selection menu instead of directly booting into live or install?

SevenBits, 2017-03-26:
@CyberTechnotrix Mac Linux USB Loader does not use the default boot menu shipping in the Kali ISO, so no, not at this time. However, Mac Linux USB Loader should automatically add boot entries for the Kali desktop and the Kali installer.

SevenBits, 2017-04-28:
Closing because this is covered by #134.

closed 2017-04-28 by SevenBits