SevenBits/Mac-Linux-USB-Loader #166: failed to boot Linux on my iMac A1394 produced by MBP running on 10.9

Ole2mail, 2017-04-01:
**Short Description of Problem:** failed to boot Linux on my iMac A1394

**Expected Behavior:** 

by making a bootable USB with already downloaded ISO image of Ubuntu to be able to plug into USB port of the iMac and boot from this key after pressing Alt key

**Actual Behavior:** 

having only EFI on the key, means, for some reason the actual ISO hasn't been copied at all. 
also the UI is so glitchy it's actually fail to produce any progress bar completely.
buttons back and forth are defective also
[image]
. 
I'm going to request moneyback on paypal within a few days if no news will comeback to me.

**Steps to Reproduce:** 
1. install "mac_linux_usb_loader.pkg"
2. execute "Mac Linux USB Loader"
3. select "create live USB"
4. provide ISO source file
5. unselect option "skip bootloader menu" and press "continue"
6. wait until the completion message will appear
7. plugin USB into iMac pressing ALT key select to boot from it
put text here

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2017-04-01:
@Ole2mail Hi there, that does not look that a screen produced by my software. Are you using a custom setup, such as rEFIt or rEFInd?

Ole2mail, 2017-04-02:
hello

here are more details about my trials:

 

- changed to HFS+ drives support

- reformated USB key to HFS+

- used following ISO image form Ubuntu site:

ubuntu-16.04.1-desktop-amd64.iso

- waited for "success" message of the UI

- the only button remain is "Previous"

 

the only folder I see on the USB key is "efi"

this is the same story as with the FAT FS drive formated.

inside efi folder I could see boot folder and inside there are 3 files

boot.efi, boot.iso, bootX64.efi

 

after booting with ALT I see no USB key drive appearance at all in the horisontal list,

the only options for boot I see 2 of my HDD partitions.

the only way the drive I'd see if I'm using reFIND, but then you know the boot error screen result.

 

thanks

Oleg

 

Sent: Saturday, April 01, 2017 at 5:19 PM

SevenBits, 2017-04-02:
@Ole2mail If your Mac will not detect your USB without the use of rEFInd, then your Mac probably does not meet the [system requirements][1]. Please ensure you are meeting all requirements.

[1]: https://www.sevenbits.io/mlul/sys-requir.html

closed 2017-04-02 by SevenBits

Ole2mail, 2017-04-02:
my iMac is capable to boot from USB key made by 
https://oemden.com/sixty-four-on-thirty-two-sfott/
or made by this
http://osxhackers.net/macpostfactor.html
it means, when the USB key is made by the tool above, I see the orange external disk once the USB plugged in and ALT powered up.
however with your product there is no disk visible at all on this iMac, similar behaviour I've got by making USB key bootable Ubuntu without using your product but just following this manual
https://business.tutsplus.com/tutorials/how-to-create-a-bootable-ubuntu-usb-drive-for-mac-in-os-x--cms-21253

so the question is really in your product - there is definitely something is missing as the icon doesn't appear in the boot menu.

Oleg

Ole2mail, 2017-04-02:
more important, I thought your tool is really direct EFI (not an UEFI aka EFI 2.x) solution for EFI32 old macintoshes.
the new macs are capable to boot Ubuntu linux (for example) directly from the regular ISO based USB key made by unetbootin, downloaded from here
https://unetbootin.github.io/
and without any "magic" hidden behind.
unetbootin is well known tool and it's available for free for quite a while already.
so for your fee it would be good to see something really improved for old macs.
thanks
Oleg