SevenBits/Mac-Linux-USB-Loader #170: Arch Linux Support

The-Wolf312, 2017-05-04:
I'm having issues getting Arch Linux to boot using this. It'd be nice to have this distribution added.

SevenBits, 2017-05-04:
> In addition I've noticed only drives formatted with MS-DOS FAT can be used on this is it possible to support much larger HFS+ partitions and or APFS partitions?

There is an experimental option in the preferences to allow installation on HFS+ formatted drives. It is disabled by default due to the fact that many distributions do not include file system drivers for HFS+ in their initial RAM discs. If your distribution does, then you may certainly attempt it.

Regarding APFS: this is so new that not even Apple fully supports it yet. It will be some time before Linux distributions include support.

> I'm having issues getting Arch Linux to boot using this. It'd be nice to have this distribution added.

It’s a possibility, though since Arch Linux is not graphical and changes so frequently, it might pose a challenge. Also it would need to support “loopback booting”, i.e. booting directly from ISO.

SevenBits, 2017-08-01:
Closing due to lack of new posts.

closed 2017-08-01 by SevenBits