SevenBits/Mac-Linux-USB-Loader #173: Booting from USB stick not working

spoudaious, 2017-05-14:
Hi,
I'm trying to create a bootable USB drive for a linux based system with a .img image, but although I have tried many softwares and procedures, I haven't succeeded yet. I am now trying your software but I'm getting this message: This USB drive cannot be used as a startup disk because OS X only recognizes startup disks with an HFS+ file system.
I'm using a MacBook Pro (13-inch, Early 2011) with OS 10.11.6.
In my other attempts, the USB drive would not be recognized on starting up with option key pressed..
Thanks in advance for any help.

SevenBits, 2017-05-14:
Mac Linux USB Loader only permits the installation of ISO images. Have you followed the steps exactly as outlined in the documentation?

spoudaious, 2017-05-14:
OK, this is not clear on the site. Can you please refund me the money then?
Thanks

SevenBits, 2017-05-14:
Please send me an email with your information.

closed 2017-05-14 by SevenBits

spoudaious, 2017-05-14:
what other information do you need?….

SevenBits, 2017-05-14:
Your email and purchase date. GitHub does not let me see this info when you comment on the issue pages.

spoudaious, 2017-05-22:
I didn’t receive the refund yet. Can you please do it today. THanks

SevenBits, 2017-05-22:
As I said, I need to know your date of purchase and either your last name or email address you used to make the purchase. You have not provided this information.