SevenBits/Mac-Linux-USB-Loader #175: USB not recognized on macbook core2duo

benninkcorien, 2017-06-01:
**Short Description of Problem:** put text here
When I press and hold ALT on boot, it does not show the USB to boot from.

**Expected Behavior:** 
Booting from USB.

**Actual Behavior:** 
Nothing happens 

**Steps to Reproduce:** 
I used mac mini to create the bootable USB (default options)
Want to install Xubuntu (latest 64bit image) on macbook core2 duo (13")

[image]
[image]

SevenBits, 2017-06-01:
Hi @benninkcorien, have you first checked the system requirements to make sure that this Mac meets all of the requirements?

benninkcorien, 2017-06-01:
I bought this Macbook in 2007 so think it meets your requirements?
I've been trying all day to get Xubuntu to install, my brain is not completely functional at this point.

SevenBits, 2017-06-01:
Please check if it has a 64-bit EFI firmware, as specified in the system requirements. A 2007 Mac might not have it.

benninkcorien, 2017-06-01:
How can I check what kind of EFI firmware my macbook has? 

Could this also be caused by the way I formatted the new SSD I put it? I can't seem to get a clean install from a DVD with OSX 10.7.2 either (I either get a folder with a question mark that's flashing, or a gray stop sign). Ugh, why do things not just work.

The SSD was first formatted as exfat on windows. When that didn't work, I formatted it as OS Journaled on a mac mini.

benninkcorien, 2017-06-01:
Update. the 32 bit is working. It's showing a Xubuntu logo and I got past the mac initial boot so I assume I can install it.

Enjoy the 5 dollar tip :) thanks for the quick replies.

closed 2017-06-01 by benninkcorien