SevenBits/Mac-Linux-USB-Loader #176: I can't update this app

gusbemacbe, 2017-06-21:
**Short Description of Problem:** 

I checked for updates and I received the new update, but I didn't find any option of updating. I check for preferences and I tried to update anyway. I use 3.4.

SevenBits, 2017-06-22:
Hi there,

The automatic updater only works when updating from 3.4.1. You will need to update manually.

closed 2017-06-22 by SevenBits

Timoofdoom, 2017-06-22:
Sorry to ask, but how do you update manually? do I go to the updates page and just download what's been updated since then?

SevenBits, 2017-06-22:
You can download the update from the link you were given when you bought the app.

Timoofdoom, 2017-06-22:
[image]

The link takes me to this page, and I've downloaded it 3 times.

SevenBits, 2017-06-22:
And what version do you get when you click that link?

Timoofdoom, 2017-06-22:
The one that I sent the other screen shot about? Says the year for it is
2012

Timoofdoom, 2017-06-22:
I'm trying Etcher at the moment, so if that works, then it's no worries.

gusbemacbe, 2017-06-22:
Timo, I contacted him. I downloaded from my email when I purchased and I
received download link. I updated to 3.4.2.

It could be 2012, but the link download always has been updated and you
just didn't reinstall and update. Download again from the download link you
received and reinstall and update.

gusbemacbe, 2017-06-22:
Etcher fails for not including grub-efi-amd64.

Timoofdoom, 2017-06-22:
Oh, man. My bad. I had mixed up the previous downloads. Yeah, I got it now. Thanks for the help!