SevenBits/Mac-Linux-USB-Loader #181: Boot does not work

mateus007, 2017-07-01:
**Short Description of Problem:** put text here
Using the software simple does not work. When I select to boot from USB, an black screen is shown, and nothing more...

I am using MacBook Pro 2016.

PS. Using dd command from ISO works beautiful, so its definitely is a problem with this App.

**Expected Behavior:** 

put text here

**Actual Behavior:** 

put text here

**Steps to Reproduce:** 

put text here

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2017-07-01:
What distribution are you trying to boot? Have you viewed the FAQ pages?

mateus007, 2017-07-01:
Ubuntu 17.04.
This could be related to issue #171

SevenBits, 2017-07-01:
That very well could be. Have you tried any of the suggestions proposed in that thread? Have you tried booting with the 'nomodeset' kernel option available?

I'm not sure why Mac Linux USB Loader would fail when you claim DD works. The underlying code in Ubuntu is the same...

SevenBits, 2017-08-01:
Closing due to lack of updates.

closed 2017-08-01 by SevenBits