SevenBits/Mac-Linux-USB-Loader #183: Can't update from 3.4.1

ajcrow93, 2017-07-22:
I cannot update from version 3.4.1 to any new version. 

When I click to install and relaunch I get: 

"An error occurred while downloading the update. Please try again later."

This has been ongoing for several weeks. 

The original download link I received when purchasing linux USB loader has expired, so I have no way of updating the App either.

SevenBits, 2017-07-23:
Hi there, could you send me the logs generated by the Sparkle updater?

Also, if you email me your original download link, I can renew it.

ajcrow93, 2017-07-25:
Hi, 

Is the log located in /Users/~/Library/Logs/SparkleUpdateLog.log  ?

Mine appears to be very sparse… 

I took a screen shot of the errors in console.app 

My original download link was this:

https://www.e-junkie.com/d/?t=1GR129672D0894323&d=v4kma&c=6rz7 <https://www.e-junkie.com/d/?t=1GR129672D0894323&d=v4kma&c=6rz7>

Thanks, 

Andrew

SevenBits, 2017-07-25:
Hi @ajcrow93, no images or resources are showing up...

I will try to renew your link.

ajcrow93, 2017-07-26:
Apologies, I replied by email and thought the attachments would upload. 
Here they are: 

[Attachments.zip](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/1175616/Attachments.zip)

closed 2017-10-02 by SevenBits