SevenBits/Mac-Linux-USB-Loader #184: Xcode indefinitly stuck on "indexing" on macOS 10.12

xapple, 2017-07-24:
I followed the build instruction exactly, though when they get to the Xcode part they become very superficial and unprecise. This is what I did:

    mkdir ~/test
    cd ~/test

    git clone https://github.com/SevenBits/Mac-Linux-USB-Loader.git
    cd Mac-Linux-USB-Loader/

    brew install cocoapods
    pod install

    open "Mac Linux USB Loader.xcworkspace"
    In XCode: Menu -> Product -> Build

The result was that when pressing "Run" or "Build" in Xcode nothing would happen, not even an error message, while the main screen perpetually displays "Indexing" as seen here:

[image]

I'm using Xcode Version 8.3.3 (8E3004b) and macOS 10.12.3 (16D32)

SevenBits, 2017-07-24:
Hi @xapple, have you tried waiting for indexing to complete before trying to build?

xapple, 2017-07-24:
Waited five hours, it was still on "indexing". But I quit and relaunched XCode and now the problem is gone. Must have been just a fluke, unrelated to `Mac-Linux-USB-Loader`.

closed 2017-07-24 by xapple