SevenBits/Mac-Linux-USB-Loader #185: Xcode complains about signing on macOS 10.11

xapple, 2017-07-24:
I tried building it on macOS 10.11.6 (15G1510) this time and at least it failed with an error message now:

     Dependency Analysis Error Group

     Check dependencies

    The operation couldn’t be completed. Unable to log in with account 'xapple'. An unexpected failure occurred while logging in (Underlying error code 1100).
     No signing certificate "Mac Development" found:  No "Mac Development" signing certificate matching team ID "***********" with a private key was found.

It would be nice if you could provide actual complete build instruction for your open-source project. Thanks!

SevenBits, 2017-07-24:
Hi @xapple, you need to disable code signing for the build to work, as you will not have access to my private signing key.

xapple, 2017-07-24:
I can understand that you wish to obfuscate the building process to increase incentive to buy the precomplied binaries of this open-source software. But do you think you could add the necessary information to the README.md after all ? Usually a product has a working "INSTALL" section...

closed 2017-10-02 by SevenBits