SevenBits/Mac-Linux-USB-Loader #186: Kali Linux - Booting live with persistence

Eno123456, 2017-07-28:
**Short Description of Problem:** I am making a live bootable persistent usb drive(32G) and  everything went fine until during the instillation process, when it was mounting ad installizing the cd-rom, it failed to find a device. 

**Expected Behavior:** 

It finds the device and continues with the instillation process. 
**Actual Behavior:** 

The instillation process halt and shows error, “The CD-ROM drive contains a CD which cannot be used for the instillation. Please insert a suitable CD to continue.”
**Steps to Reproduce:** 

Follow the instillation process normally. 

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2017-07-29:
Hi there. When you create a bootable live USB for Kali, two bootable entries will be created: one to boot into the live desktop, and the other to install. Did you select the option to install?

Eno123456, 2017-07-29:
Yes. Is that a problem?

Eno123456, 2017-07-29:
I just tried both. Still no persistence

SevenBits, 2017-07-29:
Wait, are you looking for persistence? The original post made it seem that you were having issues installing Kali...

Eno123456, 2017-07-30:
Sorry for the confusion. I'm having trouble with persistence.

Eno123456, 2017-08-01:
Hello?

SevenBits, 2017-08-01:
@Eno123456 Your original post is written as if you were having trouble installing Kali. If you are having trouble with persistence, please open a new issue, and describe exactly what is not working.

As I specify in the documentation, Kali requires special steps to set up persistence, all of which are documented [on Kali’s knowledge base](http://docs.kali.org/downloading/kali-linux-live-usb-persistence).

closed 2017-08-01 by SevenBits