SevenBits/Mac-Linux-USB-Loader #188: Launching Program Doesn't Bring Up Program Window

FromKansas, 2017-08-11:
Launch program and don't get this window.
[image]

What am I missing?

SevenBits, 2017-08-11:
Did you click on its icon in the Dock?

FromKansas, 2017-08-11:
Yes.  It brings Mac USB Linux Loader in the task bar.  All I can select is About Mac USB Linux Loader and Open.  I open my .iso file and get this. 
[image]

Drive won't show as available.

FromKansas, 2017-08-11:
I am currently using macOS 10.12.5.

SevenBits, 2017-08-11:
@FromKansas Is your USB FAT32 formatted? Have you followed all steps in the documentation?

FromKansas, 2017-08-11:
Yes I have.  My main question is why do I not get the main option window when I first launch the program?

SevenBits, 2017-08-11:
@FromKansas It should come up when you open the program, but if it doesn’t for some reason, you can show it by clicking the Dock icon or by doing View -> Welcome Screen in the menu bar.

FromKansas, 2017-08-11:
Welcome Screen under View is not an option.  
[image]

SevenBits, 2017-08-11:
Look under Window then...

FromKansas, 2017-08-11:
Not there either.  I have tried everything I can think.  Restarted the computer.  Uninstalled and reinstalled the software.  
[image]

FromKansas, 2017-08-11:
[image]

SevenBits, 2017-08-12:
@FromKansas This is odd. Did you get the app from the official website?

SevenBits, 2017-10-02:
Closing due to lack of updates.

closed 2017-10-02 by SevenBits