SevenBits/Mac-Linux-USB-Loader #190: Persistence

connred, 2017-08-25:
**Short Description of Problem:** Followed all steps to use persistence but is not working. I have set it up using the app. Then when I boot I tell it to use persistence in the enterprise menu. I tested it by placing a file on the desktop and it does not come back after a reboot.

**Expected Behavior:** 
Files will come back.
**Actual Behavior:** 
Not Files or changes stay

iswearitsberenstein, 2017-09-16:
I know this is old and you might have fixed your issue, but does the enterprise menu show up every time you boot from USB? For me, the menu comes up every time and I have to choose to turn persistence on every time (i.e. 2(to select boot options) -> 1 (to choose ubuntu) -> 0 (to launch with the selected boot options). If I don't go through all those steps every time then it won't boot with persistence. At least that is how it is for me.

Damianthirtythree, 2017-09-24:
I am not a pro at this, but my understanding is that you have to select to boot with persistence if you want to use the persistent memory. so yes. you will have select it every time.

closed 2017-10-02 by SevenBits