SevenBits/Mac-Linux-USB-Loader #191: Grant Access loop when trying to select iso

pvotava, 2017-08-27:
When selecting iso I am prompted to Grant Access, however selecting the button does not grant access.
[image]
 I have seen this issue listed before and tried your suggestions such as clearing registered devices and tried this line in terminal from prior string:
rm -R ~/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/
[image]
Just purchased today, no issue with installation. Get to here:
[image]
But selecting Grant Access does nothing except refresh with the same screen. So when I cancel I get the following message.
[image]
I formatted the drive FAT 32, here is the info:
[image]

I hope I am not missing something simple, I am not a programmer so I was counting on the app to save some time and hassle, but disappointed after prepaying, the app seems straightforward and simple but I am stuck here, and apparently not an uncommon issue? 

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2017-08-27:
Hi @pvotava. The problem is with your third screenshot -- you need to select your USB drive (possible `L M`?) in the right panel. It should be selecting this for you, but I have heard that for some users, it doesn't for whatever reason.

closed 2017-08-27 by SevenBits

pvotava, 2017-08-29:
Ahhhhh yes and success. Thank you1