SevenBits/Mac-Linux-USB-Loader #194: Kali not booting on Macbook Pro 2017 13"

vishistv, 2017-10-17:
**Short Description of Problem:** Hi, I am using the 2017 MBP 13 inches model. I have been trying to run a bootable Kali USB with persistence. I followed the steps given in the wiki. The problem that I am facing now is a blank screen with a white cursor. I hold the option key-choose EFI Boot-and then I see this. I saw that someone had already asked about this issue ( #174 ) and that he was told to refer to #158 . But here the problem is that I am not even able to access the *Advanced Boot Manager*. Kindly help me solve this issue.

**Expected Behavior:** 
Start Kali Linux

**Actual Behavior:** 
Black screen with a white cursor

**Steps to Reproduce:** 
-create Live USB using Mac Linux USB Loader
-restart the device
-hold option key
-choose EFI Boot
Blank screen with white cursor

tturn02, 2018-03-07:
I've had the same issue, man. I've been looking around the internet for hours.

SevenBits, 2018-03-08:
To be clear, you are unable to even see the boot loader? Perhaps you should try without the 'Skip the Boot Menu' option selected.

tturn02, 2018-03-13:
After some research, it seems that 2016 Macbooks just aren't compatible with Linux.

aotd1, 2019-12-19:
The same issue, the same steps to reproduce.
MacBook Pro (13-inch, 2016, Two Thunderbolt 3 ports)
Kali Linux from https://cdimage.kali.org/kali-2019.4/kali-linux-2019.4-amd64.iso
@SevenBits I tries with and without 'Skip the Boot Menu' option.

After choosing EFI Boot there is only a black screen with white cursor square on it. No blinking.

If I just flash Kali distro with `dd` it runs normally (but keyboard and touchpad drivers missing)