SevenBits/Mac-Linux-USB-Loader #195: Choose Destination Blank

raggascum, 2017-11-09:
**Short Description of Problem:** After choosing iso, next screen is choose destination to write to, blank

**Expected Behavior:** 

USB device to pick from

**Actual Behavior:** 

nothing

How do I solve this issue?

pasine, 2018-02-11:
I think the problem might be your USB device is not formatted as MS-DOS.
According to `detectAndSetupUSBs`, hfs is accepted only if you enable the experimental feature in the Preferences.

```
if ([volumeType isEqualToString:@"msdos"] ||
		([volumeType isEqualToString:@"hfs"] && acceptHFSDrives))
```
I was facing the same problem, and re-formatting my usb using FAT fixed the issue.

SevenBits, 2018-03-08:
Closing due to no new activity.

closed 2018-03-08 by SevenBits