SevenBits/Mac-Linux-USB-Loader #198: Install FULL Ubuntu, not Live CD

bradparks, 2018-01-09:
I can install Ubuntu iso on a USB, and it's the live Ubuntu.

I can boot that and run it fine. But then when I install the full Ubuntu from the Live USB on another USB, it wont boot.

I was hoping that Mac Linux USB Loader would help fix the partitions on the 2nd USB drive so it would boot correctly.

Can this be done?

Thanks,

Brad

SevenBits, 2018-01-11:
This section is for reporting bugs, not for asking general questions, email is a better choice for that.

But, what Mac Linux USB Loader is designed to do is create a bootable, live USB of Ubuntu and other Linux distributions. If you use the Ubuntu installer from within the live USB to install the full version of Ubuntu to another USB stick, Mac Linux USB Loader isn't involved as you are booted into Ubuntu.

Most Macs should be able to boot a full installation of Ubuntu just as they can boot a live USB created with Mac Linux USB Loader -- by holding down the Option/Alt key at boot and selecting the correcting drive.

I hope this helps.

closed 2018-01-11 by SevenBits

bradparks, 2018-01-11:
I'm mostly uniformed about this, but can't the mac only boot EFI enabled boot USB's ? Isn't that one of the main things that Mac-Linux-USB-Loader does? 

and fwiw, I did buy the 5$ commercial version of this to support you, and am not asking for you to implement this feature, I just thought that's what it did ;-) I had already created a bootable USB using some blog post on the web, and thought this would make it easier to do that, which it does, but thought it would possibly "fix" the boot record of an ubuntu full install to a USB drive, which is what I really wanted to do, so I can run a full version of Ubuntu from my mac, which is a work mac, and I can't change the hard drive on it (ie dual boot).

Thanks!