SevenBits/Mac-Linux-USB-Loader #202: No options shown in "Choose Destination" window

sarky-de, 2018-04-30:
**Short Description of Problem:**
The window for selecting an USB drive is empty even though a drive is present and accessible via Finder. Selecting "Refresh" will still result in an empty window.
[image]

**Expected Behavior:** 
USB flash drive shows up in window and can be selected.

**Actual Behavior:** 
USB flash drive does NOT show up in window and therefore cannot be selected.

**Steps to Reproduce:** 
macOS Sierra 10.12.6

SevenBits, 2018-05-01:
Have you verified that your USB drive meets the requirements as detailed in the program documentation?

sarky-de, 2018-05-02:
Issue can be closed, thanks!

> Chances are that while the operating system says the USB drive is formatted as FAT, it might be formatted as another variant of FAT not supported by UEFI, such as exFAT. To resolve this issue, the drive should be reformated (WARNING: make sure there is no irreplacable data on the drive beforehand) to FAT32 (FAT) using Disk Utility.

closed 2018-05-02 by sarky-de