SevenBits/Mac-Linux-USB-Loader #203: Issues with Build

JeanJacques-Rousseau, 2018-05-13:
**Short Description of Problem:** 

I get an error from attempting a build with xcode

**Expected Behavior:** 
Successful build. 
**Actual Behavior:** 

error: could not read data from '/Users/Jean-JacquesRosseau/Mac-Linux-USB-Loader/findercontext/Info.plist': The file “Info.plist” couldn’t be opened because there is no such file.

** BUILD FAILED **

The following build commands failed:
	ProcessInfoPlistFile build/Release/findercontext.appex/Contents/Info.plist findercontext/Info.plist
(1 failure)

**Steps to Reproduce:** 

1. git clone https://github.com/SevenBits/Mac-Linux-USB-Loader.git
2. Run Pod Install 
3. cd mac-linux-usb-loader
4. xcodebuild CODE_SIGN_IDENTITY="" CODE_SIGNING_REQUIRED=NO

The error then arises. 
[image]
[image]

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

orlleite, 2018-05-16:
I'm trying to use this app too. And I have similar problem.

You have to removed findercontext.appex from building inside xcode.
After I have another problem related to pod files. I removed and readded in linked frameworks and libraries. And finally I could build. BUT, you need to signing you code. You can run but you can't drive your usb flash. Without it, you can't have the context for make changes of an usb drive, so no copying files. That is apples rules, AFAIK. I don't know if there is a way to get ride of that.

amlestin, 2018-05-19:
Compile one of these instead. I used the latest successfully.

https://github.com/SevenBits/Mac-Linux-USB-Loader/releases