SevenBits/Mac-Linux-USB-Loader #204: Cant boot after writing to the drive

33masterman33, 2018-07-12:
**Short Description of Problem:** I have a respin of linux mint and the iso works fine but when I use mac linux usb loader and try to boot it skips the programs grub and goes to the one in the ISO but after a little while of waiting for it to turn on it says (initramfs) Unable to find a medium containing a live file system
**Expected Behavior:** 

Boot into the OS

**Actual Behavior:** 

(initramfs) Unable to find a medium containing a live file system

**Steps to Reproduce:** 

Not quite sure how to have someone else reproduce it

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

LoganDark, 2018-08-25:
I get the same issue with a 4GB SanDisk USB drive and a Mid-2015 MBP.

LoganDark, 2018-08-25:
[image]

SevenBits, 2018-08-25:
@LoganDark, could you try using a different USB drive?

LoganDark, 2018-08-25:
Unfortunately, no. This is my only one.

spadazz, 2018-10-13:
I just had the exact same error, after flashing on my 4gb sandisk stick a linux mint 19 cinnamon iso.  
I tried flashing both with and without the option "This iso lacks an EFI enabled kernel".  
In both cases I got the initramfs error.  
I also tried in nomodeset option, no luck.  
I used the same iso without any problem in a virtualbox guest.  
I really need to reinstall a 2009 imac with linux.  
What mint distro is known to work with this app ??

LoganDark, 2018-10-13:
@spadazz 2009 imacs can install without MLUL IIRC

spadazz, 2018-10-14:
@LoganDark . I don't know wtf is this "MLUL IIRC".  
I even tried to google it, but no luck.

spadazz, 2018-10-14:
Indeed I tried to to change usb stick.  
Since the 4 GB sandisk was giving ACPI error, I disabled acpi from boot options and mint started.  
I didn't try to install it because I'm not on the actual mac to reinstall now.  
Then I retried to flash another stick sandisk cruzer 8GB with etcher, that previously didn't work at all (with a different stick).
It booted without any complaint or need to disable boot options.  
So buy a 10$ stick of a reliable brand like sandisk or samsung and you should be ok.

LoganDark, 2018-10-14:
MLUL = This program (Mac Linux USB Loader)
IIRC = If I Recall Correctly

And my USB drive *is* a Sandisk.

I hate it when people like you try to give advice.

spadazz, 2018-10-15:
I basically said the same thing that Sevenbits did.
Because I verified it.  with 2 different sandisk, both working.
You first commented  almost 2 months ago.  
No need to wait further for a single usb stick that don't work.
I don't care if you've had a bad day, and you want to lash out on me.
But maybe someone else will read thru the comments and find a viable solution.

LoganDark, 2018-10-15:
...