SevenBits/Mac-Linux-USB-Loader #206: Where are the recent pre-built versions?

LoganDark, 2018-08-24:
Okay, no matter what I do in Xcode, this stupid pile of shit won't build, so my only hope at this point is to try to find a pre-built version... I spent TEN MINUTES waiting for CocoaPods to clone its retarded 2-gigabyte git repo just to learn that this thing won't build without a $100 developer certificate, so no thanks.

But all the downloads I can find here... are from six years ago.

So what can I do? Where can I find a pre-built version?

robotoad, 2018-08-24:
I don't know how much support you'll get with that attitude, but you can purchase a prebuilt copy [here](https://www.sevenbits.io/mlul/).

LoganDark, 2018-08-24:
I'm sorry, but frankly I'm pissed because I've been sinking hours of my time into this with no luck. And seriously, you have to purchase this shit? No thanks. Why the fuck is it open source if I have to buy it?

closed 2018-08-24 by LoganDark

robotoad, 2018-08-24:
You don't have to buy it, but I doubt anyone will want to help you now.

LoganDark, 2018-08-24:
Hence why I closed the issue. I'm just pissed. This irritates me to no end.

SevenBits, 2018-08-24:
@LoganDark In order to build the software, you need Cocoapods installed and you need to run Cocoapods before building the project. This is a very commonly used dependency manager for Apple platforms. If you want to build software from source, you need to have a basic understanding of how this is done. I’m sorry if this is frustrating. As for the $100 certificate issue... Apple is pretty much moving towards requiring all applications to be signed by a registered developer and be sandboxed in order to run. Thus, certain features of this application require users to have a certificate in order to build. Again, this is just how it is. If you don’t have the time or inclination to figure this out, you can pay a small fee (price of a cup of coffee) to get a compiled version that is ready to go.

Also, in the future, I would encourage you to adopt a better attitude when asking for help on an online forum.