SevenBits/Mac-Linux-USB-Loader #207: "Choose Destination" is empty, but "Detected potentially eligible volume"

LoganDark, 2018-08-24:
When I want to create a Live USB, the destination box is empty:

[image]

Xcode says:

[image]

It's a 4GB SanDisk Cruzer USB stick. I erased it using Disk Utility. What's going on here?

SevenBits, 2018-08-24:
HFS+ support can be enabled in the preferences, but it is experimental as Linux distributions have never had good support for it. That is unlikely to change now that Apple is abandoning this file system.

FAT32 is the only one that will work. This is enforced by the underlying system firmware; I can’t do anything about it.

LoganDark, 2018-08-25:
Okay, thanks.

It finally worked, I succeeded in booting into Ubuntu 18.04.1 Desktop, but the loading screen shows for a while before it disappears and shows a black screen featuring the absolute latest in BusyBox technology (/s) and initramfs complaining it can't find a live FS.

I'll open another issue.

LoganDark, 2018-08-25:
Instead of posting a new issue, I posted a comment on #204, they have the same issue.