SevenBits/Mac-Linux-USB-Loader #211: stuck on enterprise

aikendad, 2018-10-08:
Using Mac Linux USB Loader version 3.4.6 I've setup a usb drive with Ubuntu 18.04. When booting it stops at "welcome to enterprise - version 0.4.1". This is on a mac mini late 2014.

Found a closed (due to inactivity) request with the same problem and an "enterprise.cfg" file was requested. I have found no such file on my usb drive. See attached screen shot.

[image]
Any help would be appreciated.

aikendad, 2018-10-09:
Okay, I found how to make the enterprise.cfg file visible and here is a text file version of it
[enterprise.cfg.txt](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/2462586/enterprise.cfg.txt)

Toovedama, 2019-06-02:
Hi! I have the exact same problem, but I can't understand your solution. Am I supposed to write that text in one of the documents?

Honestly this downloading of Linux on my mac is a lot more difficult than the wizards say it is, mainly because they use words I don't know and explanations I can't understand. I am not necessarily stupid, but I have other interests than computers.