SevenBits/Mac-Linux-USB-Loader #214: USB Loader Destination issue

darginsky, 2018-12-03:
I just purchased your software for use on my Macbook Air 2014, OSX Mohave. However, when I reach Choose Destination screen, no drives are populated, even after "refresh". Please help.

Also, It will not download distros, the View In-Progress button just opens a blank partial window.

tas23466, 2019-01-25:
You should try to erase the USB with Disk utility and select the MS-DOS format, then re-run the program...

gerkov77, 2019-02-23:
After purchasing, and installing your product, it couldn't open my USB Drive as destination. I have erased my USB drive with with Disk Utility with  MS-DOS format selected. MAC LINUX USB Loader still can't select my USB Drive.

jahbini, 2019-04-25:
The logic in file `Mac-Linux-USB-Loader/SBAppDelegate.m` at lines 322 is:
```BOOL isValidFormat = [volumeType isEqualToString:@"msdos"];
					isValidFormat |= acceptHFSDrives && [volumeType isEqualToString:@"hfs"];
					isValidFormat |= acceptHFSDrives && [volumeType isEqualToString:@"apfs"];
```
The `accetpHFSDrives` boolean is set on the application preference pane.

[image]