SevenBits/Mac-Linux-USB-Loader #223: Boots with persistence, but no persistence actually there

mattyount28, 2019-10-24:
**Short Description of Problem:** 
I am trying to get Kali Linux with persistence on my MacBook with a 320GB external USB drive (by seagate). Everything in the setup process works as expected, no errors or anything. But, no files remain after a reboot as expected.
**Expected Behavior:** 
I create a file on the desktop, do a reboot, and the file remains.
**Actual Behavior:** 
I create a file on the desktop, do a reboot, and the file is gone.
**Steps to Reproduce:** 
I reformatted my drive in FAT. I selected create live USB with the Mac Linux USB Loader menu. I then selected the persistence manager and added persistence with the max storage (4GB) and used the default name (casper-rw). I can start Kali and it does detect the persistence, so I go to advanced, select Kali (as opposed to Kali Installer), and select persistence and then click 0 to boot. I do this every time I boot, but any files that I save are not actually saved, and every time I boot to the drive, all of my settings and files are restored to defaults.

abuquet, 2020-01-20:
Same issue on my macbook pro late 2013 with a 64Go USB key and kali

alexanderslash, 2020-10-15:
Same issue here with the latest Ubuntu distribution ubuntu-20.04.1-desktop-amd64. The files are gone after reboot although it recognises (and I select) persistence.

awgrass, 2021-04-06:
> Same issue here with the latest Ubuntu distribution ubuntu-20.04.1-desktop-amd64. The files are gone after reboot although it recognises (and I select) persistence.

In Ubuntu 20.04 the persistent file/persistent partition name has been officially changed from "casper-rw" to "writable". Since Mac Linux USB Loader did not integrate those changes yet, you have to manually rename the persistent file ("casper-rw") in the root of the stick to "writable". After that, restart your Mac and proceed by setting the persistent option in the boot menu and start ubuntu (2 - 1 - 5 - 0).  Persistence is now enabled.